epics-base/pvAccessCPP #198: EPICSchat.exe detected as potentially malicious

Atix95, 2024-09-23:
Hey everybody,

when compiling EPICS-base the epicschat.exe from the example folder is run by default.

Since we are using a new antivirus software this is detected as potentially malicious because it establishes an upstream connection to the IP address 192.168.1.187. Currently, we are compiling epics-base without the epicschat.exe as a workaround, as we do not require it for our usecase.

Has anybody else had this problem? And does anybody know what is behind this IP address?

I have added a screenshot of the file analysis from the Cisco client (Cisco is our antivirus software).

If you need any further information please let me know!

Best regards
Michael

[image]

anjohnson, 2024-09-23:
Hi,

The example epicschat.cpp program in the EPICS Base `modules/pvAccess/examples` source directory gets compiled when the pvAccessCPP module is built, but it shouldn't be being run automatically at all, even if you run our self-tests after the build process. When that program runs it starts a PV Access server, which listens for connections from PV Access clients. Any connections that are made to it would have been requested by the client, not the epicschat server.

IP addresses in the range 192.168.0.0/16 are not routable across the global internet and can only be accessed within your local network from other addresses in the same subnet (192.168.1.187 is an address in your subnet). If you start the epicschat program on a local host it will broadcast a series of UDP packets to the local subnet, but shouldn't be making any connections of its own. None of the network streams you show above appear to be using the PV Access protocol which uses ports 5074 and 5075. 

What version of EPICS Base did you compile, and where did you download it from? I'm not seeing anything in the information you provided which would come from a normal EPICS system.

- Andrew

mdavidsaver, 2024-09-23:
> What version of EPICS Base did you compile, and where did you download it from?

I think this is the first question to answer.

I hope that your answer will be a clone of this github repository, or a download from either epics-controls.org or epics.anl.gov .

> I have added a screenshot of the file analysis from the Cisco client (Cisco is our antivirus software).

The streams shown look like DHCP and netbios (aka. windows file sharing).  Is your listing complete?

I also don't see any executable names mentioned.

Atix95, 2024-09-24:
Hi everybody,

thanks for the fast responses!

I am sorry if some things were missing or unclear. This is the first time I am opening an issue and I am not familiar with networks so I could only pass on the information from our IT.

> What version of EPICS Base did you compile, and where did you download it from?

We cloned the latest release from the epics-base github repository [7.0.8.1](https://github.com/epics-base/epics-base/releases/tag/R7.0.8.1)

> The streams shown look like DHCP and netbios (aka. windows file sharing). Is your listing complete?

The lisiting is complete.

> The example epicschat.cpp program in the EPICS Base modules/pvAccess/examples source directory gets compiled when the pvAccessCPP module is built, but it shouldn't be being run automatically at all, [...]

> IP addresses in the range 192.168.0.0/16 are not routable across the global internet and can only be accessed within your local network from other addresses in the same subnet (192.168.1.187 is an [...]

Thanks for the explanation. I believe it is possible that this is detected as a false positive by Cisco. I will talk with our IT about this.

Best regards
Michael

mdavidsaver, 2024-10-06:
> > The streams shown look like DHCP and netbios (aka. windows file sharing). Is your listing complete?
> 
> The lisiting is complete.

Then I don't think that epics-base is involved.  Please re-open if you come by additional information.

closed 2024-10-06 by mdavidsaver